microsoft/pxt-adafruit #579: The Same project, simulator doesn't work under "BLOCKS".

Emma-sunny, 2017-12-20:
**Repro Step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Create New Project and Switch to _{}JAVASCRIPT_
- Copy the following code into content panel and run it 
"
_let spinit = false;
let pixels = light.createStrip();_

    _input.buttonA.onEvent(ButtonEvent.Click, () => {
        spinit = false;
        pixels.clear();
    })_
_control.runInBackground(() => {
    pixels.setPixelColor(0, Colors.Blue);
    while (spinit) {
        pixels.move(LightMove.Rotate, 1);
        loops.pause(250);
    }
})_
_spinit = true;
for (let i = 0; i < 5; i++) {
    loops.pause(1000);
    pixels.setPixelColor(0, Colors.Blue);
}_
"
- Then switch back to _BLOCKS_ and run it

**Expected result:**
After switching back to "_BLOCKS_", it should work as like working on "_{}JAVASCRIPT_".
[image]

**Actual Result:**
[image]

Helen09, 2018-03-07:
**Actual result:**
After clicking "BLOCKS", it shows error
 
[image]

samelhusseini, 2018-04-25:
We no longer support the colors enum in blocks. Use the color value instead like: 0xff0000

closed 2018-04-25 by samelhusseini