microsoft/pxt-adafruit #581: How to force update?

ladyada, 2017-12-20:
on my win 7 desktop w/firefox i still get 

> A Blocks / JavaScript code editor for the Adafruit Circuit Playground Express.
> Adafruit Circuit Playground Express version: 0.10.13
> Microsoft MakeCode version: 1.3.6
> C++ runtime version: master

how can i force it to update? I tried 'reset' to no avail :)

pelikhan, 2017-12-20:
Could open the JavaScript console and share the output on a full reload?

ladyada, 2017-12-20:
> Logging system initialised at Wed Dec 20 2017 12:49:25 GMT-0500 (Eastern Standard Time)
> common.js:195:17
> Loading failed for the <script> with source “blob:https://makecode.adafruit.com/92cd40c7-d330-4a81-9318-45040d2d9277”.
> makecode.adafruit.com:1
> The Application Cache API (AppCache) is deprecated and will be removed at a future date.  Please consider using ServiceWorker for offline support.
> makecode.adafruit.com
> Download the React DevTools for a better development experience: https://fb.me/react-devtools
> main.js:49721:9
> unreachable code after return statement
> [Learn More]
> main.js:15201:4
> Browser: firefox 57.0 on windows
> pxtlib.js:1219:13
> Logging system initialised at Wed Dec 20 2017 12:49:26 GMT-0500 (Eastern Standard Time)
> common.js:195:17
> Logging system initialised at Wed Dec 20 2017 12:49:31 GMT-0500 (Eastern Standard Time)
> common.js:195:17
> Ignoring non-existent field NUM in block neopixel_colors

[image]

pelikhan, 2017-12-20:
Could you also dump the network tab data? I wonder why it is trying to get that initial URL.

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F1214161%2F34220966-6a0bdffc-e584-11e7-855e-b3477da2e3b9.png&data=02%7C01%7Cjhalleux%40microsoft.com%7C437e23d0425c4f00e73208d547d23865%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636493890575631209&sdata=1LkSk%2FhShfidgS%2FUxU10xRODHizb9%2BKkIEnX%2B0L79kM%3D&reserved=0>

ladyada, 2017-12-20:
do you know an elegant way to extract the data? its in a table >.<

pelikhan, 2018-04-03:
Is this still an issue?

ladyada, 2018-04-03:
i never resolved it, so i just deleted all browser data which is one way of fixing the problem :D

abchatra, 2018-06-07:
No repro

closed 2018-06-07 by abchatra