microsoft/pxt-adafruit #586: The simulator didn't display clearly on firefox

yanlxu, 2017-12-29:
Repro steps:
• Navigate to https://makecode.adafruit.com/beta on **firefox**
• Drag blocks like the followed picture
• Drag block "on shake" to the content panel
[image]

Expected result:
The simulator should display clearly

Actual result:
The simulator didn't display clearly 

[image]

microsoftsam, 2017-12-29:
What version of Firefox?

microsoftsam, 2017-12-29:
Please add browser details in the future: Browser version and Operating System.

yanlxu, 2018-01-02:
@microsoftsam, the version of Firefox is 57.0.3, the Operating System build: \\winbuilds\release\RS_ONECORE_WEBPLAT_STAGE_DEV3\17058.1004.171227-1700\x86fre

Helen09, 2018-03-09:
The issue still repro on the latest build: 17607.1011.180305-1700 on firefox(version: 58.0.2)
[image]

Helen09, 2018-05-10:
Closed as not repro

yanlxu, 2018-05-10:
close as not repro

closed 2018-05-10 by yanlxu