microsoft/pxt-adafruit #588: Error shows when switch block "onEvent" example code from "{  }JAVASCRIPT" to "BLOCKS"

Helen09, 2018-01-04:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta 
• Navigate to "ADVANCED" -> "CONTROL" -> "onEvent(0, 0, function( ){  })"
• Drag this block and right click to look its help document
• Copy the example code to "{ }JAVASCRIPT" and then click "BLOCKS"

**Expected result:**
 There should switch to "BLOCKS" successfully
 **Actual result:**
Share Project: https://makecode.com/_e9Viwa7pT1Fm
Error shows when switch block "onEvent" example code from "{  }JAVASCRIPT" to "BLOCKS"
[image]

Helen09, 2018-04-25:
The example has changed to the below code and the bug still repros.
const pixelLighter = 22 
let pixels = light.createStrip()

control.runInParallel(() => { 
     for (let i = 0; i < 2; i++) { 
        pause(1000)          
     control.raiseEvent(pixelLighter, i) 
   } 
})

control.onEvent(pixelLighter, 0, () => { 
     pixels.setPixelColor(0, 0xff0000) 
})

control.onEvent(pixelLighter, 1, () => { 
     pixels.setPixelColor(1, 0x0000ff) 
})

Helen09, 2018-04-25:
I get the example:
"ADVANCED" -> "CONTROL" -> block "on event from 0 with value 0"

samelhusseini, 2018-04-25:
Issue here is still the createStrip  method.