microsoft/pxt-adafruit #596: After adding block "reset", project can't be published

Helen09, 2018-01-11:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/
• Drag blocks like below picture
• Click "SHARE" to publish this project

**Expected result:**
 The project should be published successfully
**Actual result:**
This project can't be published for adding block "reset"
[image]

samelhusseini, 2018-04-25:
Can't repro anymore on /beta. Please try again and reopen if still repro.

Would be great in this scenario if you could add a JS copy of the code used so we can debug further.

closed 2018-04-25 by samelhusseini

Helen09, 2018-04-25:
I validate this bug, it not repro on Edge, but it still repro on **Firefox, Chrome & IE**
**Firefox**: 59.0.2
**Chrome**: 66.0.3359.117
**IE**: 11.1.17134.0

**The JS copy of the code:**
forever(function () {
    light.showRing(
        "red red red red red red red red red red"
    )
    control.reset()
})

Helen09, 2018-04-25:
**The beta version info:**
Adafruit Circuit Playground Express version: 1.1.9
Microsoft MakeCode version: 3.8.18
C++ runtime version: v0.2.2

Helen09, 2018-04-25:
The detail is as below .gif
[image]

Helen09, 2018-04-26:
Could you help reopen this bug?@Sam

samelhusseini, 2018-04-26:
It's the same root cause as this: https://github.com/Microsoft/pxt-adafruit/issues/702