microsoft/pxt-adafruit #597: Plug-in-to-USB goes to bootloader not program?

ladyada, 2018-01-18:
some instructors seem to expect that if you plug into a USB port, the program will run, not UF2

https://forums.adafruit.com/viewtopic.php?f=58&t=127805&p=644900

we could update UF2 or just instruct people? @mmoskal

pelikhan, 2018-04-03:
It's going to be very difficult to change this behavior without confusing existing users.

abchatra, 2018-06-07:
Some teachers like the behavior we have today as they don't have to ask kids to go to bootloader mode for the first flash and program will run as soon as you flash. Changing this is going to be hard. May be we should better document the behavior. @ganicke What is the right place to document the behavior?

ganicke, 2018-06-07:
I see a few places to add a note for this:

https://makecode.adafruit.com/device
https://makecode.adafruit.com/device/usb
https://makecode.adafruit.com/device/usb/troubleshooting

ladyada, 2018-06-07:
fyi i dont care to change it, just reporting from forums!

pelikhan, 2018-06-07:
I've subscribed to the MakeCode forum now.

mmoskal, 2018-06-07:
Hold on! We just need to enable the storage module in CPX. Then you can flash without hitting reset and we can require double-reset to go to bootloader (in case your program crashes badly and the storage module stops working; this shouldn't happen if you're not using C++).

ladyada, 2018-06-07:
i would luuuuv the storage module - did that get made!!?

pelikhan, 2018-06-07:
Working on undusting it: https://github.com/Microsoft/pxt-adafruit/pull/740

ladyada, 2018-06-07:
sweeet i can test when its bumped - let me know. i have tons of .wav playing code ive written over the years, i know that would be a massively popular ability if available :)

pelikhan, 2018-06-07:
Would you have some engineering cycles to add support playing .wav in CODAL? That piece of the puzzle is still missing :(

ladyada, 2018-06-07:
i could take a look when the file system access stuff is ready, and if DMA stuff is all handled for me!