microsoft/pxt-adafruit #598: A block "Middle C" has two displaying forms.

Emma-sunny, 2018-01-19:
**Repro Step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Drag "_set item to 0_" from "_VARIABLE_" and put it into "_forever_"
- Then put "_Middle C_"(Music) into "_set item to ‘0’_ " and click "_Middle C_"

**Expected Result:**
[image]

**Actual Result:**
[image]

abchatra, 2018-06-07:
@samelhusseini is this worthwhile to fix for the release?

samelhusseini, 2018-06-07:
Fix is being integrated.

closed 2018-06-07 by samelhusseini