microsoft/pxt-adafruit #604: analogWrite to pads doesnt work?

ladyada, 2018-01-24:
hiya trying to generate a PWM signal on the pads does not seem to do anything?

[image]

also, relatedly, A8, A9 and A10 should not appear here as they are internal sensors and its weird to have them be outputs! :)

[image]

pelikhan, 2018-01-24:
The pin map is defined here: https://github.com/Microsoft/pxt-adafruit/blob/master/libs/circuit-playground/device.d.ts#L34 . Any other we should remove?

ladyada, 2018-01-24:
just A8, A9, and A10 should be removed from that list
A0 doesnt have a timer so PWM is hard to do but you can do true analog writes

jedgarpark, 2018-02-11:
I'm trying to do analogWrite to A0, but this seems like it currently doesn't output any voltage when you send 1023 or any other values.

pelikhan, 2018-02-11:
Is it just A0?

jedgarpark, 2018-02-11:
that's the only one I'm trying because i want a real analog signal, not PWM like the other pins.

pelikhan, 2018-02-12:
Looking into this.

pelikhan, 2018-02-13:
And the results are in:
- [ ] A0: lock
- [ ] A1: ok
- [ ] A2: ok
- [ ] A3: ok
- [ ] A4: lock
- [ ] A5: lock
- [ ] A6: lock
- [ ] A7: lock

pelikhan, 2018-04-03:
A fix is in beta. @ladyada We ended up restricting PWM out to A1 and A2. Use 3 PWM pins would lock the device and pins A4, A5, A6, A7 lock the device as well because of an mbed issue (see https://github.com/lancaster-university/codal-circuit-playground/issues/5). Maybe we can get some cycles from your engineers to help fix this issue.

ladyada, 2018-04-03:
ok according to my docs, only A1, A2 A3 and A6 have PWM out, so A1 and A2 are good!

ladyada, 2018-04-13:
@pelikhan heya related Q, i want to try this and also test out a customer issue https://forums.adafruit.com/viewtopic.php?f=58&t=133680 but i can only update to 1.0.28 adafruit CPX release which is from ~2 weeks ago and it looks like some stuff was released since. is there a way to 'force' makecode to upgrade to the latest?

[image]

pelikhan, 2018-04-13:
The easiest trick to defeat caching is to use incognito/inprivate/anonymous mode is your browser... or add ?v=djebeizizkfb to the url where
djebeizizkfb is generated by banging the keyboard :)

We also stage our next update on https://makecode.adafruit.com/beta . That version has Michal’s fixes to support analogWrite on A0.

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F1214161%2F38715933-262aafe4-3eac-11e8-8ce9-2532046dacff.png&data=02%7C01%7Cjhalleux%40microsoft.com%7Cff5def44f27942358f2c08d5a0f1ac0c%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636591882190386917&sdata=zHQ2PY0XjAosW%2FNV4KQsDo3sr17JIzZQ5Y4bk9CAXS8%3D&reserved=0>

abchatra, 2018-06-07:
This bug seems to be fixed.

closed 2018-06-07 by abchatra