microsoft/pxt-adafruit #606: Prompting information is not displayed under "High Contrast on"

Emma-sunny, 2018-01-25:
**Repro Step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Click "_More..._" and choose "_High Contrast On_"
- Select "_VARIABLE_" or "_FUNCTIONS_"

**Expected Result:**
[image]

**Actual Result:**
"_Variable_":
[image]

"_Functions_":
[image]

Helen09, 2018-03-28:
This issue also repro on Edge

Emma-sunny, 2018-04-26:
Close as fixed

closed 2018-04-26 by Emma-sunny