microsoft/pxt-adafruit #611: Block "Console log..." doesn't work.

Emma-sunny, 2018-01-26:
**Repro Step:**
- Navigate to [https://makecode.adafruit.com/beta](url)
- Create new Project and copy the following code into content panel:
_let list: string[] = []
let newList: string[] = []
newList = ["pellow", "newWorld", "About", "568789"]
for (let value of list) {
    console.log(value)
}_

**Expected Result:**
1. Block "_Console log..._ " should show all string into the following area.
[image]

2. Button "_Show console Simulator_" should be displayed below the Simulator.
[image]

**Actual Result:**
_Mac:_
[image]

_Windows:_
[image]

Helen09, 2018-03-28:
This bug still repro on Chrome&Safari on Mac
On other browsers, it shows as below:
[image]

Emma-sunny, 2018-04-11:
Closed as fixed

closed 2018-04-11 by Emma-sunny