microsoft/pxt-adafruit #616: Editor is not displayed correctly when user right-click mouse and click backspace

shunzhan, 2018-02-01:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta
• Switch to "BLOCKS" mode
• Click any block(eg: show ring) in the editor and pressing the backspace

**Expected result:**
 Block should be displayed correctly and block should be able to be dragged

**Actual result:**
On firefox: The block overlapped
[image]
On chrome&edge&safari: The block can not be dragged
[image]

samelhusseini, 2018-04-25:
Woah, the size of the field editor in firefox is a little out of wack.

abchatra, 2018-06-07:
I can't reproduce this

abchatra, 2018-06-07:
Reopen if you can

closed 2018-06-07 by abchatra