microsoft/pxt-adafruit #618: The Function name didn't display on the content panel when the "High Contrast" is on

yanlxu, 2018-02-01:
Repro steps:
• Navigate to https://makecode.adafruit.com/beta
• Click the "High Contrast On" under setting "More..."
• Click the "FUNCTIONS" in the editor
• Make a Function named "test"

**Expected result:**
1. It should display "Make a Function" when click the "FUNCTIONS", just like on **Edge&IE**
2. The Function name "test" should be displayed on the content panel

**Actual result:**
1. It didn't display "Make a Function" when click the "FUNCTIONS" on **Chrome&FireFox**
2. The Function name "test" didn't display on the content panel on all browser

**Chrome&FireFox:**
[image]

**Edge&IE:**
[image]

All browser:
[image]

samelhusseini, 2018-04-23:
Similar to the issue with the color field. We need to get rid of the tron look I think.

samelhusseini, 2018-04-25:
Looks like a dupe of https://github.com/Microsoft/pxt-adafruit/issues/606

closed 2018-04-25 by samelhusseini