microsoft/pxt-adafruit #619: Is cap touch supported?

ladyada, 2018-02-01:
heya folx

a person noted that there's no blocks for capacitive touch, but i vaguely remember being asked about it. 
https://forums.adafruit.com/viewtopic.php?f=58&t=130273
i looked in both the beta and release site but only see onClick() type things, not any FreeTouch stuff

pelikhan, 2018-02-01:
The ``input.pin*`` access the pins as capacitive touch buttons. The ``pins.pin**`` access the pins as digital/analog pins.

It's a bit confusing - we should probably document that further.

pelikhan, 2018-02-01:
Source of truth: https://github.com/Microsoft/pxt-adafruit/blob/master/libs/circuit-playground/device.d.ts

pelikhan, 2018-02-01:
I do think we still have some calibration issues on the cap touch and Joe was looking into that as well.

pelikhan, 2018-02-01:
FYI https://github.com/lancaster-university/codal-core/issues/3

ladyada, 2018-02-01:
[image]

these are the blocks you're referring to?

pelikhan, 2018-02-01:
All the blocks under "input" are going to be capacitive. It's an unfortunate naming convention - we should have called them "touchA1" to be clearer since A1 could also be used as a digital button.

ladyada, 2018-02-01:
i think yeah especially if you're going to use the words 'pressed' and 'clicked' its confusing because i'd think they would be a tact button
perhaps a new block that is 'was pressed' would be good. also 'button pin' should be renamed 'touch pin'

pelikhan, 2018-02-01:
Let me discuss this with the team. We need to be careful about not breaking existing scripts when renaming things. I'll drop the plan in this issue so that everyone agrees.

eat-sleep-code, 2018-02-01:
@pelikhan   This is the actual code I was executing (see four click handlers).   Seems to work fine in the emulator but does not register the touch when deployed to my Circuit Playground Express:

```
let gameColors: number[] = [Colors.Pink, Colors.Green, Colors.Blue, Colors.Yellow]
let gameCorners: string[] = ["topLeft", "topRight", "bottomLeft", "bottomRight"]
let currentGameCorners: string[] = []
let currentGameColors: number[] = []
let playedGameCorners: string[] = []
let keysPressed = 0

// Capture capacitive touch inputs
input.pinA5.onEvent(ButtonEvent.Click, function () {
    playedGameCorners.push("topLeft")
    keysPressed += 1
    music.playTone(220, music.beat(BeatFraction.Half))
})

input.pinA2.onEvent(ButtonEvent.Click, function () {
    playedGameCorners.push("topRight")
    keysPressed += 1
    music.playTone(247, music.beat(BeatFraction.Half))
})

input.pinA6.onEvent(ButtonEvent.Click, function () {
    playedGameCorners.push("bottomLeft")
    keysPressed += 1
    music.playTone(262, music.beat(BeatFraction.Half))
})

input.pinA1.onEvent(ButtonEvent.Click, function () {
    playedGameCorners.push("bottomRight")
    keysPressed += 1
    music.playTone(294, music.beat(BeatFraction.Half))
})

// Get a random corner from the corner list
function getRandomCorner() {
    return gameCorners[Math.floor(Math.random() * gameCorners.length)];
}

// Get a random color from the color list
function getRandomColor() {
    return gameColors[Math.floor(Math.random() * gameColors.length)];
}

// Render a random corner in a random color
function displayCorner(corner: string, color: Colors) {
    switch (corner) {
        case "topLeft":
            light.setPixelColor(0, color);
            light.setPixelColor(1, color);
            break;
        case "topRight":
            light.setPixelColor(9, color);
            light.setPixelColor(8, color);
            break;
        case "bottomLeft":
            light.setPixelColor(3, color);
            light.setPixelColor(4, color);
            break;
        case "bottomRight":
            light.setPixelColor(5, color);
            light.setPixelColor(6, color);
            break;
        default:
            break;
    }
    loops.pause(500)
    light.setAll(0)
    loops.pause(500)
}

// Render one complete round
function displayRound() {
    let randomColor: number = getRandomColor();
    let randomCorner: string = getRandomCorner();
    currentGameColors.push(randomColor)
    currentGameCorners.push(randomCorner)
    for (let _i = 0; _i <= currentGameCorners.length - 1; _i++) {
        displayCorner(currentGameCorners[_i], currentGameColors[_i]);
    }
}

// Play 50 rounds 
for (let rounds: number = 0; rounds <= 50; rounds++) {
    displayRound()
    while (keysPressed < currentGameCorners.length) {
        // WAITING FOR PLAYER INPUT...
    }
    if (currentGameCorners[keysPressed - 1] != playedGameCorners[keysPressed - 1]) {
        keysPressed = 0
        currentGameColors = []
        currentGameCorners = []
        playedGameCorners = []
        music.playSoundUntilDone(music.sounds(Sounds.Wawawawaa))
        for (let _i2 = 0; _i2 <= 5; _i2++) {
            light.setAll(Colors.Red)
            loops.pause(200)
            light.setAll(0)
            loops.pause(200)
        }
    }
    else {
        keysPressed = 0
        playedGameCorners = []
        loops.pause(1000)
        if (rounds == 50) {
            music.playSoundUntilDone(music.sounds(Sounds.PowerUp))
            light.showAnimation(light.sparkleAnimation, 2000)
        }
    }
}

```

pelikhan, 2019-03-13:
I am thinking about renaming the "input.pin *" names to "input.touch *" to make it clearer that those are touch events ("input.pin *" will still be there but hidden for appcompat.

ladyada, 2019-03-13:
i think thats wise!

pelikhan, 2019-03-14:
"touch *" added in https://github.com/Microsoft/pxt-adafruit/commit/f402c718ee1ed79556ad7f1994b9f7e2d6ced3c7

pelikhan, 2019-03-14:
I'm closing this issue as fixed. Please reopen if something else is missing. Once we release the new editor, i will be able to update all our tutorials to migrate to "pin *"

closed 2019-03-14 by pelikhan