microsoft/pxt-adafruit #630: Add more description to expanded cards on Home Page

Jaqster, 2018-02-09:
Hi Galen,
Can you add some more text description to the Tutorials, and Example cards on the Home Screen?
Right now they look very bare.  Maybe just add things like:
Level: Beginner / Intermediate / Advanced
Description: In this project / step-by-step tutorial, you will build a program that will show an animation and play a sound when you shake the Circuit Playground Express.
Materials: (if this is a Make projects)

ganicke, 2018-02-09:
Sure, I'll give it a try. Don't think card rendering supports more than one paragraph so we might not get the coolness your looking for.

pelikhan, 2018-02-09:
We should define these properties as metadata and special render them.

ganicke, 2018-02-09:
Yea, add some more stuff to the 'schema' for ``cardType: "gallery"`` and render in separate \<p\> elements.

```json
{
  "name": "Hot Potato",
  "url":"/projects/hot-potato",
  "imageUrl":"/static/cp/projects/thumb/hot-potato.jpg",
  "description": "Pass on the potato before you get burned! Lights and sounds let you know if it's still cool or if you just got scorched."
  "materials": ["Tape", "Paper sheet"],
  "level": "Beginner",
  "cardType": "gallery"
}
```

ganicke, 2018-02-09:
What does @samelhusseini think?

pelikhan, 2018-02-09:
We should start with level probably.