microsoft/pxt-adafruit #632: [UWP] Device locks when using serial

pelikhan, 2018-02-09:
This program locks when connected to the windows app
```
loops.forever(function () {
    light.graph(Math.randomRange(0, 10), 0)
})
```

mmoskal, 2018-02-09:
Was it broken with WebUSB? Or was it non-functional before WebUSB?

pelikhan, 2018-02-09:
It works https://makecode.adafruit.com/

rachel-fenichel, 2018-02-09:
Bugreporting for a friend:
This is exposed in the "Loudness necklace" tutorial with this code that the tutorial says to write:
```
loops.forever(function () {
    light.graph(input.soundLevel(), 255)
})
```

"Froze within a second of starting with USB attached.  Disconnecting USB and running off of a battery caused the code to run fine without downloading it again.  Plugging back into USB immediately caused the lockup again.  

Hardware in use:

- Lenovo laptop T430 running Fedora 26 Linux
- Adafruit Circuit Playground Express (latest version)
Tutorials for Siren, Magic Wand, and Clap Lights all worked fine on this hardware setup with the USB cable both attached and disconnected.  (Disconnected running on lithium battery.)"

Possible priority bump since it's in a tutorial?

pelikhan, 2018-02-09:
Could you ask if this was using the /beta app? @mmoskal

rachel-fenichel, 2018-02-10:
How do they know if they're using the beta?  Is the difference whether they were at https://makecode.adafruit.com/#editor vs https://makecode.adafruit.com/beta/#editor, or something else?

pelikhan, 2018-02-10:
It's the only way to tell right now. Or go to gearwheel -> About and report the version number. Thanks!
[image]

mmoskal, 2018-02-10:
It's possible this is broken under Linux also for HID not only WebUSB. If this is the case, we will need to apply a fix like the one in https://github.com/Microsoft/pxt-common-packages/pull/223

rachel-fenichel, 2018-02-10:
Version info: 

[image]

mmoskal, 2018-02-10:
So I guess we do need to avoid venting HID packets into space - apply https://github.com/Microsoft/pxt-common-packages/pull/223 uniformly also to HID. 

This would require changes in the Win 10 though, so that data logging continues to work - I guess the changes are only on the WebApp side, and similar to the ones we need for WebUSB.

pelikhan, 2018-02-10:
@rachel-fenichel could you try https://makecode.adafruit.com/beta#editor (v0.1.13)?

jafo408, 2018-02-10:
Hello there,

I am the "friend" on whose behalf Rachel reported this.  I just now signed up for my GitHub account at her request so I could comment directly.

I'm confirming the beta#editor mentioned above works for this code.  Screenshot of the version that works.  

[image]

Thank you for all your help on this.  Original bug identification goes to my 9 year old daughter.  :-)

pelikhan, 2018-02-10:
You can thank your daughter for us! Is that her first reported/fixed bug? Well done!

[makecode_version_beta]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F36319174%2F36057996-f0cebca2-0dcb-11e8-8095-6128952dd9ed.png&data=04%7C01%7Cjhalleux%40microsoft.com%7C134ccec48fe64d45c74508d57032fa8c%7Cee3303d7fb734b0c8589bcd847f1c277%7C1%7C0%7C636538286615129125%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwifQ%3D%3D%7C-1&sdata=RIK3mY%2BD8Zq7x9csxELAuphQliV84sRZ2lgk7B2PrOQ%3D&reserved=0>

jafo408, 2018-02-10:
I will definitely pass that along!  Yes, this was indeed her first bug find in someone else's code.  She's an active Scratch programmer at school and was excited when she could make a Magic Wand using the same method.  I told her earlier this evening that she had found a real bug and she got very excited. I'm sure your kind comments from the actual developers will be on proud display in her elementary school classroom Monday morning!

abchatra, 2018-06-07:
This is fixed

closed 2018-06-07 by abchatra