microsoft/pxt-adafruit #639: how to create an extension with event handlers

jallwine, 2018-02-19:
I'm interested in creating an extension for pxt-adafruit that provides blocks that respond to events that I define, similar to the "on infrared received" block. I've read some of the pxt documentation but the instructions don't work or aren't complete. I'm looking for some pointers on where to start, not for someone to implement this for me. This is a mockup of one of the blocks I'd like to make:

[image]

I'd like to be able to listen to the serial port and if that block exists and a specific message is detected over serial, run the code within that block with an argument that contains data from the serial message. Any pointers on how to get started would be appreciated. I was able to follow the instructions [on the pxt-adafruit repository](https://github.com/Microsoft/pxt-adafruit) to get a local pxt target running, but I quickly run into errors when I try to follow the ["Building your own package" instructions](https://makecode.com/packages/getting-started) (the setting up your workspace link is dead and Step 1 isn't very clear).

The goal is to be able to interface with a Minecraft mod I developed called [SerialCraft](https://github.com/AllwineDesigns/SerialCraftMod). I've written an [Arduino library](https://github.com/AllwineDesigns/serialcraft-arduino) that makes it easy to write Arduino sketches that interface with the SerialCraft mod, but it would be great to be able to use the makecode.adafruit.com editor to get comparable behavior. I'd like to be able to implement projects like [this one](https://www.instructables.com/id/Creeper-Detector/) with only the makecode.adafruit.com editor and the circuit playground express.

jallwine, 2018-02-19:
This is the error I get when following the [Setting up a workspace](https://makecode.com/cli#setting-up-a-workspace) instructions.

```
$ mkdir adafruit
$ cd adafruit
$ pxt target adafruit
$ pxt serve

Using target PXT/adafruit with build engine codal
  Target dir:   /Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-adafruit
  PXT Core dir: /Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core
Going to /Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-adafruit
No sim/tsconfig.json; assuming npm installed package
Starting server in /Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-adafruit
With pxt core at /Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core
starting local ws server at 3233...
---------------------------------------------

To launch the editor, open this URL:
http://localhost:3232/#local_token=15155374-4d2c-4266-7c16-6dbe16ee30f5&wsport=3233

---------------------------------------------
opening http://localhost:3232/#local_token=15155374-4d2c-4266-7c16-6dbe16ee30f5&wsport=3233
INTERNAL ERROR: Error: ENOENT: no such file or directory, open '/Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-adafruit/libs/base/pxt.json'
    at Object.fs.openSync (fs.js:646:18)
    at Object.fs.readFileSync (fs.js:551:33)
    at readJson (/Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core/built/nodeutil.js:260:26)
    at readPkgConfig (/Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core/built/nodeutil.js:266:14)
    at Object.resolveMd (/Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core/built/nodeutil.js:467:19)
    at readMd (/Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core/built/server.js:682:28)
    at Server.<anonymous> (/Users/jallwine/work/AllwineDesigns/Blog/CreeperDetector/adafruit/node_modules/pxt-core/built/server.js:865:22)
    at emitTwo (events.js:126:13)
    at Server.emit (events.js:214:7)
    at parserOnIncoming (_http_server.js:602:12)
    at HTTPParser.parserOnHeadersComplete (_http_common.js:117:23)
```

abchatra, 2018-06-07:
Is this question still relevant?

pelikhan, 2018-06-07:
We should add an example to the playground.

pelikhan, 2018-06-15:
@jallwine we are working on the UART blocks. We also have more documentation on blocks at https://makecode.com/playground.

pelikhan, 2019-03-21:
@jallwine you do not need a local build to create a package anymore. If you follow these instructions, https://makecode.com/blog/github-packages you can do it all from the editor.

Try out https://makecode.adafruit.com/beta for the latest bits.

closed 2019-03-21 by pelikhan