microsoft/pxt-adafruit #645: Can't see which block is hovered on "High Contrast" mode

Helen09, 2018-02-24:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta
• Click "LIGHT" and then hover on each block
• Click "setting" and then click "High Contrast On" and hover on block again

**Expect result:**
When hover on one block, this block should have a color shade change or other
[image]

**Actual result:**
After clicking "High Contrast On" and then hover on blocks, there is no color change, so user can't see which block is selected
[image]

samelhusseini, 2018-04-23:
Right, tron look again.

pelikhan, 2018-04-23:
Time to rip it out.

closed 2018-05-04 by samelhusseini