microsoft/pxt-adafruit #649: Maker having issue with IR transmit/receive

ladyada, 2018-03-02:
https://forums.adafruit.com/viewtopic.php?f=58&t=128057&p=655458#p655202

https://makecode.com/_XMbFRr4WoFqW

> When I load this into either of the two CPX that came with my educator's kit, I can get the normal behavior I expect by shaking the CPX and the lights lit up the ring. When its full, tilt up and stuff happens.
> 
> If you press "A" it will run the code with the IR Send. If I try it with the same code on a second device, I sometimes see the response animation... so the IR is sent and received. However, after I try the IR Send (A button), I usually start to hear slowed/distorted sounds (shake and the normal tone sounds gravelly) and degraded performance (e.g. it seems not to respond consistently to new "shake" inputs).

abchatra, 2018-06-07:
@riknoll Can we try out this IR scenario to see if this reproduces?

pelikhan, 2019-03-13:
Clock issue.

closed 2019-03-13 by pelikhan