microsoft/pxt-adafruit #654: Download error from app

Jaqster, 2018-03-06:
In my session today, there was a user who could not get the Download to work from the App.
He was running a Lenovo Thinkpad, we tried multiple boards/cables/ports to no avail.  Save worked fine.
[image]

abchatra, 2018-06-07:
I can't repro this. But we should give an option to save file if a device is not found. @guillaumejenkins Is it easy for Friday release?

guillaumejenkins, 2018-06-07:
I think right now we show the download dialog only if the error is **not** "Device not found". I can change that logic to show the dialog on every error. I think keeping the "Device not found" message is good. I should be able to do that for the release.

PR #745 (2018-06-08): Add troubleshooting when UWP app fails to flash device
Closes #654

Requires https://github.com/Microsoft/pxt/pull/4453 to work

@ganicke please review the .md file
closed 2018-06-08 by pelikhan