microsoft/pxt-adafruit #655: Synth tempo inconsistent, fast BPM

jedgarpark, 2018-03-07:
Longer songs play well in the simulator, but when played back from the real CPX the tempo seems to be inconsistent from measure to measure.

In the example of the James Bond Theme I transcribed,
 https://makecode.com/_Fe3gExag7ixK 
(press the button A to play it) the intro starts of pretty close to the tempo in the simulator, but then it rushes the rest of the song like crazy.

It's not vital that the tempo match exactly between the simulator and the real hardware, but it's important that the tempo stays consistent on either device.

Additionally the 'set tempo to __ bpm' seems to be incorrect even in the simulator. I haven't done a proper test to figure out the exact factor, but seems to be roughly 3x too fast. I set a song that should be 120bpm to 35bpm in MakeCode to prevent it from racing through it too fast.

abchatra, 2018-06-07:
Likely clock issue.

pelikhan, 2019-03-12:
Looks like the tempo is ok but the sound is not really good.

closed 2019-03-12 by pelikhan