microsoft/pxt-adafruit #656: serial blocks

jallwine, 2018-03-07:
Why are there no serial communication blocks like the microbit version of make code? This would be a fantastic way to interface over USB with programs running on a computer. Is this in the road map?

pelikhan, 2018-03-09:
The CPX for MakeCode supports HF2, https://github.com/Microsoft/uf2/blob/master/hf2.md, but not serial. HF2 allows to interact with WebUSB or the windows app.

jallwine, 2018-03-10:
Are there any blocks for sending arbitrary data using WebUSB? What windows app are you referencing?

samelhusseini, 2018-04-25:
Windows app: https://www.microsoft.com/en-us/store/p/makecode-for-adafruit/9pgzhwsk0pgd?SilentAuth=1

abchatra, 2018-06-07:
You need windows app and console.log block will send data to the app.

closed 2018-06-07 by abchatra