microsoft/pxt-adafruit #660: Shaking simulator works differently with clicking "SHAKE" button

Helen09, 2018-03-09:
**Repro steps**:
• Navigate to https://makecode.adafruit.com/beta
• Drag blocks as below .gif
• Click "SHAKE" button on simulator few times
• Shake simulator few times

**Expect result**:
When shaking simulator, it should work as clicking "SHAKE" button
[image]

**Actual result**:
When shaking simulator, it only works once
[image]

abchatra, 2018-04-23:
Dupe of https://github.com/Microsoft/pxt-adafruit/issues/626.

closed 2018-04-23 by abchatra

Helen09, 2018-04-25:
This issue is not same as https://github.com/Microsoft/pxt-adafruit/issues/626 , so please reopen it.

samelhusseini, 2018-04-25:
I get what's going on. Won't fix.