microsoft/pxt-adafruit #661: Block comment can't be selected

Helen09, 2018-03-14:
**Repro steps**:
• Navigate to https://makecode.adafruit.com/beta
• Click "LIGHT" to drag any block like "show ring" and put it under forever
• Right-click to add comment for "show ring"
• Try to select the comment and deleted

**Except result**:
The comment should be selected 
**Actual result**:
The comment can't be selected 
[image]

samelhusseini, 2018-04-25:
Fixed in /beta

samelhusseini, 2018-04-25:
Please verify.

closed 2018-04-25 by samelhusseini

Helen09, 2018-04-25:
This bug has been fixed, but the comments shows differently on Edge&Chrome

**Chrome: 66.0.3359.117**
[image]

**Edge:**
[image]

Adafruit Circuit Playground Express version: 1.1.9
Microsoft MakeCode version: 3.8.18
C++ runtime version: v0.2.2

samelhusseini, 2018-04-25:
You are looking at two different versions of the editor. One is the old style of comments and the other is the new style.. If you reload the previous editor in Chrome, you'll notice they're the same.

Helen09, 2018-04-26:
Ok, I got it. Thank you~