microsoft/pxt-adafruit #673: The content on "JavaScript" content pane is not clear

Helen09, 2018-03-29:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta#editor
• Click "ADVANCED" -> "+EXTENSIONS" ->"keyboard"
• Then switching from "BLOCKS" mode to "{}JAVASCRIPT" mode
• Drag blocks in ToolBox "GAMEPAD" to JavaScript content pane

**Expect result:**
The content on content pane should be clear
[image]

**Actual result:**
The content on content pane is not clear
(After switching to 'High contrast on' mode and then switching to 'High contrast off' mode again, it can shows clearly)
[image]

samelhusseini, 2018-04-25:
Well. So there is an issue here in that when you switch back from high contrast we don't re-load the theme for coloring...

What you're describing is a poor choice in the color of the category, perhaps we can do something to change the color slightly knowing that it's a dark toolbox, but ultimately it's a choice of color of the package. 

Won't fix for now, but will fix this issue with not returning the original coloring of text.

samelhusseini, 2018-04-25:
Fixed in PR.

closed 2018-04-25 by samelhusseini