microsoft/pxt-adafruit #674: After switching to "BLOCKS" mode, the program didn't run

Helen09, 2018-03-29:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta#editor
• Copy the below code on "JavaScript" pane and the click button A

let strip = light.createStrip()
input.buttonA.onEvent(ButtonEvent.Click, () => {
    strip.showAnimation(light.cometAnimation, 2000)
})

• Switch to "BLOCKS" mode and click button A again

**Expect result:**
After clicking button A, the program should run in "BLOCKS“ mode as in 'JavaScript' mode
**Actual result:**
After switching to "BLOCKS" mode, the program didn't run
[image]

samelhusseini, 2018-04-25:
There's something funny going with the decompilation. @riknoll 

```
let strip = light.createStrip()
input.buttonA.onEvent(ButtonEvent.Click, () => {
strip.showAnimation(light.cometAnimation, 2000)
})
```

is decompiled into 
[image]

and then back to
```
let strip: light.NeoPixelStrip = null
input.buttonA.onEvent(ButtonEvent.Click, function () {
    strip.showAnimation(light.cometAnimation, 2000)
})
strip = light.createStrip(pins.A1, 0)

```

hence why it's not running.

samelhusseini, 2018-04-25:
Same issue here: https://github.com/Microsoft/pxt-adafruit/issues/626

samelhusseini, 2018-04-25:
Another here: https://github.com/Microsoft/pxt-adafruit/issues/603

samelhusseini, 2018-04-25:
Another dupe here: https://github.com/Microsoft/pxt-adafruit/issues/593

pelikhan, 2018-06-06:
https://github.com/Microsoft/pxt-adafruit/commit/952869c15cbc33b16dd7f42c7c0f97da51e1f6ea

closed 2018-06-06 by pelikhan