microsoft/pxt-adafruit #681: These two blocks don't have same selection menus

Helen09, 2018-04-11:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta#editor
• Click "ADVANCED" -> "PINS" and then drag blocks like below picture
• Click the selection menu
**Expected result:**
These two blocks should have same selection menus
**Actual result:**
These two blocks don't have same selection menus
[image]
[image]

pelikhan, 2018-04-11:
By design. Pin hav different capabilities.

[1]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F23466737%2F38602888-9efcd4bc-3d5b-11e8-9c18-0d6f0f669f70.PNG&data=02%7C01%7Cjhalleux%40microsoft.com%7Ce6f18ee6d10d431dadf208d59f823886%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636590304010314362&sdata=OcWzx7S%2FjvMqHVY4kXDlqUwTlB9F5ddh1yh07Eu5P4A%3D&reserved=0>
[2]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F23466737%2F38602893-a30f9cc4-3d5b-11e8-9450-08531e6dcdad.PNG&data=02%7C01%7Cjhalleux%40microsoft.com%7Ce6f18ee6d10d431dadf208d59f823886%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636590304010324371&sdata=dUk3iIJzZJ8Skr6AfCbuc%2FukM1cYtzsfF5ObuZ2vjQ4%3D&reserved=0>

Helen09, 2018-04-12:
But if user wants to use these two blocks together, the block "console log value" don't have A0、SCL、SDA...

samelhusseini, 2018-04-25:
Closing by design.

closed 2018-04-25 by samelhusseini