microsoft/pxt-adafruit #682: Documentation or pointers for server installation?

brennen, 2018-04-11:
Howdy,

I'm trying to put together some quick instructions for our users to install a standalone / development instance of this project on a headless system (ec2 instance, DigitalOcean Droplet, that sort of thing).

It's clear enough how to do this on a desktop box, but with the current release I'm running into breakage when I attempt to run `pxt serve --cloud` on an Ubuntu system without a desktop environment.

Getting errors like the following:

```
~/pxt-adafruit $ pxt serve --cloud
Could not determine target's pxt-core location, falling back to default: /home/ubuntu/pxt
Using target adafruit with build engine codal
  target: v /home/ubuntu/pxt-adafruit
  pxt-core: v /home/ubuntu/pxt
Invalid accessToken format, expecting something like 'https://example.com/?access_token=0abcd.XXXX'
building node_modules/pxt-core...
[run] cd node_modules/pxt-core; npm run build
INTERNAL ERROR: Error: Cannot autolaunch D-Bus without X11 $DISPLAY
```

Grepping the source I note that it's looking for an environment var like `PXT_ACCESS_TOKEN='https://example.com/?access_token=0abcd.XXXX'`, but setting that doesn't seem to get me any further (and this doesn't seem to be documented).

Any pointers?  Thanks!

pelikhan, 2018-04-14:
There was a bug with key tar. Please pull and try again. The "invalid access token" should go away now.

brennen, 2018-04-16:
Thanks - will have a look.

brennen, 2018-04-16:
That bit does seem fixed.  On an Ubuntu 16.04 system, I'm still getting:

```
$ pxt serve --cloud
Using target adafruit with build engine codal
  target: v1.1.4 /home/ubuntu/pxt-adafruit
  pxt-core: v3.9.3 /home/ubuntu/pxt
INTERNAL ERROR: Error: Cannot autolaunch D-Bus without X11 $DISPLAY
```

riknoll, 2018-04-16:
`pxt serve` isn't meant for hosting; it's really only meant for development purposes. If you're trying to host a headless deployment (even if for development) you can use `pxt staticpkg` instead to generate a static version of the website and then host it with any old HTTP server. That might be an easier solution if `pxt serve` isn't working out.

brennen, 2018-04-16:
Yeah, right on. That's in fact what I'm looking for in the server case.  Thanks.

brennen, 2018-04-17:
So on the Linux systems I'm attempting to build on with `pxt staticpkg` (all Debian derivatives, so probably a bit behind the times), I get a slew of compiler errors.  What build environment is actually expected?

riknoll, 2018-04-17:
You need everything here for starters:
https://github.com/lancaster-university/codal#manual-installation

samelhusseini, 2018-04-25:
Looks like this issue is resolved, please reopen if not.

closed 2018-04-25 by samelhusseini