microsoft/pxt-adafruit #689: Autocompletion of music.playground in monaco is incorrect

abchatra, 2018-04-16:
music.playSound(Sounds.PowerUp) throws errors. Type music.playSou and tab

samelhusseini, 2018-04-25:
@riknoll this is once again broken because the parsing of eg changed, this time I'm not able to get it to work correctly for both the toolbox and autocomplete. 

I tried: eg: music.sounds(Sounds.PowerUp)
but that fixes it for autocomplete, but the toolbox now parses it as a string, and adds quotes around it. 

I tried adding it to sound.defl="music.sounds(Sounds.PowerUp)" but it shows up as an empty string in autocomplete in that case and a the string representation still in the toolbox. 

Why are the two entry points still different? ie: why am I seeing inconsistencies between the two cases.

Reference block: pxt-common-packages/melodies::playSound

abchatra, 2018-06-07:
@riknoll Is this possible to fix for friday release?

riknoll, 2018-06-07:
I swear I am going to just remove the e.g. parsing from PXT. Tired of fixing it.

pelikhan, 2018-06-07:
Would not be a bad idea. If you do, we need some kind of compiler check to make sure we don't break a ton of blocks.

closed 2018-06-07 by samelhusseini