microsoft/pxt-adafruit #706: Limit to supported servo pins

MrTrick, 2018-05-01:
The Makecode IDE only permits controlling servos on pins `A1` and `A2`.

Adafruit's [pinout diagram](https://learn.adafruit.com/assets/47156) shows that pins `A2`,`A3`,`A6`,`A7' support PWM.  *(Admittedly servo control is not strictly the same as 'PWM')*

I am not sure how servo control is implemented internally, but is seems like an arbitrary limitation so raising as bug; Can all CPX pins support `servoWrite`?

*[See also this Adafruit forum post](https://forums.adafruit.com/viewtopic.php?f=58&t=134873)*

abchatra, 2018-06-07:
@pelikhan @mmoskal any reason about this limitation?

pelikhan, 2018-06-07:
https://github.com/lancaster-university/codal-circuit-playground/issues/5

pelikhan, 2019-03-14:
It used to lockup. We need to revisit this in no-mbed. @jamesadevine

pelikhan, 2019-03-19:
@ladyada looks like the pinout drawing is missing the PWM curly for A1

[image]

ladyada, 2019-03-19:
the person who made this diagram for us has disappeared and they used a custom font so we can't seem to edit the PDF. will make a note of it but nothing much we can do

pelikhan, 2019-03-19:
You could just wing it with gimp? :)

ladyada, 2019-03-19:
gimp doesnt edit pdfs (the original file/download is a pdf)
i added a note

pelikhan, 2019-03-19:
Might be lucky with InkScape for a PDF...

ladyada, 2019-03-19:
im totally out of time this week :D if someone wants to shave this yak, the PDF is here
https://learn.adafruit.com/adafruit-circuit-playground-express/downloads

pelikhan, 2019-03-19:
Limition here: https://github.com/lancaster-university/codal-samd/blob/d12c428f49c9b2c3cccf14841b9c82e7926934d6/src/pwmout.c#L218