microsoft/pxt-adafruit #708: Photon color bug

Jaqster, 2018-05-02:
I think this is a bug...  The Photon Set Pen Hue block isn't working as expected.
1.) The Hue color picker colors don't seem to match.  When I select what looks like yellow (173), it actually shows up as blue.
[image]
2.) When I drag in an actual color block, that doesn't work either
[image]

pelikhan, 2018-05-02:
(2) is expected since you're passing an RGB color as the hue. (1) looks like a valid bug.

Jaqster, 2018-05-02:
Okay.  For #2, is there any way we can throw an error or otherwise alert folks that the types are different?

pelikhan, 2018-05-02:
The RGB color will most likely be out of 0..255 range but we typically silently ignore incorrect values. Both API take number, nothing to do here.

Jaqster, 2018-05-02:
Alrighty, @ganicke maybe we can just be more explicit in the docs (https://makecode.adafruit.com/reference/light/set-photon-pen-hue) - "you can't use the RGB Color block here"

ganicke, 2018-05-03:
Hue vs. RGB note added in #709.

ganicke, 2018-05-03:
Color should be RED <---> RED, I think?

abchatra, 2018-06-07:
Yikes, I can repro this 😞

pelikhan, 2018-06-07:
Fixed.

Jaqster, 2018-06-08:
I'm still seeing the hue bug Peli...