microsoft/pxt-adafruit #711: The text is selected when opening "Extensions" window

Helen09, 2018-05-04:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta#editor
• Click "EXTENSIONS" and then close this window
• Repeat the second step again

**Expected result:**
The text should not be selected when opening "Extensions" window
**Actual result:**
The text is selected when opening "Extensions" window
[image]

samelhusseini, 2018-05-04:
I don't repro. What browser / OS?

Helen09, 2018-05-04:
The machine build is 17134.1.180410-1804.rs4_release.amd64fre.enterprise_en-us_vl
Edge version: 42.17134.1.0

abchatra, 2018-06-07:
Doesn't repro anymore as we have a new UI

closed 2018-06-07 by abchatra