microsoft/pxt-adafruit #723: Set Variable to not compatible with color (R,G,B) assignment on Win10 App

TheKitty, 2018-05-21:
**Describe the bug**
A clear and concise description of what the bug is: In Win10 App, MakeCode for Circuit Playground Express, the set variable block is not recognizing the color combiner block as returning a long, it thinks it returns a string. Pops triangle error due to type mismatch that should not exist.

Error "Argument of type 'string' is not assignable to parameter of type 'number'.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Create forever loop
2. Add Set variable to block
3. Add LIGHT -> red green blue values to number assignment
4. See picture, full text of statement is set item to red 255 green 255 blue 255. 
5. Program things the color number generator returns a string when it is a number which combines the colors to one long such as 0xFFFFFF for 255, 255, 255

**Expected behavior**
Expected numeric variable item to be assigned combined value of red, green, blue numbers for use in a NeoPixel color assignment later

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 
[image]

**Desktop (please complete the following information):**
 - OS: Win10 Spring 2018
 - Browser N/A (Win10 app, IE underneath?)
 - Version Adafruit Circuit Playground Express 1.1.9, MakeCode 3.8.18, C++ runtime 0.2.2

**Additional context**
Add any other context about the problem here.

pelikhan, 2018-05-21:
This issue is fixed in /beta.

abchatra, 2018-06-07:
fixed

closed 2018-06-07 by abchatra