microsoft/pxt-adafruit #727: wifi sockets

ladyada, 2018-05-26:
i was cleaning up my filesystem and noticed i had an old idea/design for a CPX bolt-on that would add WiFi

in short - i think PXT w/ internet would be neat. BUT the lost cost ESP family doesnt have native USB or a lot of pins. i suppose we could make an mbed bootloader for ESP32 but i'm still seeing some issues with that chip (e.g. no neopixel support, i2c iffy). maybe having some sort of 'bolt on' kit for CPX/microbit could work where it gives you a sockets pipe thru an ESP8266?

we could use seesaw to give us a I2C <-> UART interface (0.50 cent chip with a huge amount of memory for buffering UART data) and we could use AT commandset? side benefit is you can use it to load new firmware on over I2C too. [there's a somewhat working AT commandset for ESP right now](https://github.com/espressif/ESP8266_AT). but its old, and isn't awesomepants.
[cell modules have a nicer AT commandset that could be ported to ESP](https://portal.3gpp.org/desktopmodules/Specifications/SpecificationDetails.aspx?specificationId=1515)

something else?

pelikhan, 2018-06-08:
We have seesaw supported in MakeCode, does this open the door to all these improvements?

ladyada, 2018-06-08:
hardware wise, it would just do the underlying data couriering. that's not the hard part. hard part is having socket/https supported in makecode/codal natively 
basically, was there every a plan to IoTify MakeCode? i dont think its anytime soon but if it were, i have a use for it :)

pelikhan, 2018-06-08:
I wonder if James looked into it at some point. I think this is probably something that community/partner should build as a package.

pelikhan, 2018-06-08:
Speaking of seesaw, you should now be able to port seesaw-based CircuitPython libraries to makecode fairly easily.

ladyada, 2018-06-08:
kk - we have a couple seesaw-based feathers but crickit is the big'un.

pelikhan, 2019-03-13:
This idea morphed into the Esp32SPI bridge?

ladyada, 2019-03-13:
right now yep all we have is this SPI based code https://github.com/ladyada/Adafruit_CircuitPython_ESP32SPI
you need 4 pins (SPI) + 1 busy state and 1 reset pin

ladyada, 2019-03-13:
(id like to look at i2c slave for ESP32 but given the prediliction for the chip to go off into space, that may not be possible)