microsoft/pxt-adafruit #732: Custom RGB not working in BLOCK mode

dalirium, 2018-05-30:
I cannot generate a custom color by just using block...
Your own 'Reference' material has the example.

visit:
https://makecode.adafruit.com/reference/light/rgb

Then follow the block sample to create a custom Yellow color (Cannot be done in block)
Luckily, the javascript is also provided and works fine:

**let strip = light.createStrip()
let myYellow = light.rgb(255, 228, 181)
input.buttonA.onEvent(ButtonEvent.Click, () => {
    strip.setAll(myYellow)
})**

But when you switch to block you get an error 
[image]

_"Argument of type 'string' not assignable to parameter of type 'number'."_

and the code stops working.

If i switch back to Javascript code here's what I see instead of what I had initally entered:

**let myYellow = 0
let strip: light.NeoPixelStrip = null
input.buttonA.onEvent(ButtonEvent.Click, function () {
    strip.setAll(myYellow)
})
strip = light.createStrip(pins.A1, 0)
myYellow = light.rgb("" + 0, "" + 190, "" + 210)**

All I want is to easily create a unique color using RGB

Hovdekorp, 2018-06-05:
I have the exact same problem in both the online version and the APP.

abchatra, 2018-06-07:
fixed in https://makecode.adafruit.com/beta . Fix will be released to live soon.

closed 2018-06-07 by abchatra