microsoft/pxt-adafruit #736: The website will crashed when opening example code in JavaScript mode

Helen09, 2018-06-06:
**Repro steps:**
• Navigate to https://makecode.adafruit.com/beta
• Scroll down and open example code 
• For example: "Music" -> "Beat visualizer" and then click "Open Example"
• Click "{ }JavaScript" to switch to JavaScript editor

**Expected result:(Edge&Firefox&IE)**
The example code can be opened in editor successfully
**Actual result:(Chrome)**
The website will crashed when opening example code in JavaScript mode
[image]

Helen09, 2018-06-06:
**Additional info:**
Chrome Version: 67.0.3396.6

samelhusseini, 2018-06-06:
I haven't been able to repro this on Chrome Version 67.0.3396.79 on Mac. 

Are you able to consistently repro? If so this is concerning.. @abchatra

samelhusseini, 2018-06-06:
Looks like Chrome fixed this issue on 67.0.3396.79

Closing this.

closed 2018-06-06 by samelhusseini