microsoft/pxt-adafruit #755: [Edge] Block rendering completely broken by undo / redo

guillaumejenkins, 2018-06-08:
Doesn't seem to happen on Chrome

1. Have a stack of 2 blocks inside an event handler
2. Drag them both out
3. Put them back in
4. Ctrl-Z a few times until both blocks are not connected anymore
5. Ctrl-Shift-Z to redo and add the blocks back into the event handler
6. ????????????????????? !))"&"?/!"(*(!?@

[image]

samelhusseini, 2018-06-08:
This isn't a browser compat issue. 

This is an undo stack issue. The pxt issue was closed as it was tracked in the wonder repo, so you're welcome to keep this open but here's a link to it.
https://github.com/Microsoft/pxt/issues/3535

guillaumejenkins, 2018-06-08:
Closing as dupe

closed 2018-06-08 by guillaumejenkins