microsoft/pxt-adafruit #765: CPX locks when using IR, light level, sound level

pelikhan, 2018-06-09:
```
I Love the makecode environment for Circuit Playground Express.

I'm creating a ton of projects to use with a summer programming camp.

I thought up a fun project where each camper would create a Firefly/Lightning bug that had a random color.
The bug would flash that color after a random delay. 
But in addition it would send out its color hue via IR so that any other bug in view would see the blink, and reply with a blink in the same color.

The code works great, except eventually each bug will freeze showing a color it was just sent (after 3 or 4 blinks).
At that point, no other part of the program works until it's reset.

The program is pretty simple, so I'm wondering if it's my bug, or some issue with the firmware.

Here is the project: https://makecode.com/_dihJ1gC29Ye2
You need to load it into 2 or 3 CPX's
```
https://forums.adafruit.com/viewtopic.php?f=64&t=136682&p=677199#p677199

pelikhan, 2018-06-15:
Here are repoes:
```
input.buttonA.onEvent(ButtonEvent.Click, function () {
    network.infraredSendNumber(42)
})
music.setVolume(20)
forever(function () {
    music.playTone(262, music.beat(BeatFraction.Half))
    music.rest(music.beat(BeatFraction.Half))
})
forever(function () {
    pins.LED.digitalWrite(false)
    pause(100)
    pins.LED.digitalWrite(true)
    pause(100)
})
```

pelikhan, 2018-06-15:
lock
```
forever(function () {
    network.infraredSendNumber(0)
})
forever(function () {
    light.showAnimation(light.rainbowAnimation, 500)
})
```

gearsincorg, 2018-07-12:
Hi... I'm the original poster...

Glad this issue got reported here.  

The IR network is a great feature for kids to write games that interact with each other.  Two boys in my group wanted to create a baseball came with pitcher and batter.  Unfortunately, this bug got in their way.  They tried synching with lights but it wasn't very effective.    

Am willing to help code/test etc.  Where would I start looking in codebase?

gearsincorg, 2018-08-01:
Further info:

The main trigger event for this bug is Sending an IR code.  Receiving a code does not cause the same problem.

It seems like the act of sending an IR code is changing a hardware timer somewhere, but not setting it back again.  This same timer must be also used to generate sounds.  Sounds seem to be slowed down by a factor of 8 or so.

gearsincorg, 2018-09-11:
@pelikhan 
Where do I look to see the actual low level code that is implementing the Send IR Code.  
I'm an embedded developer, but I just can't find my way around all the repos and components to find the nitty/gritty code for this.

jedgarpark, 2018-11-12:
I've got a possibly related issue happening with a receiving CPX locking up after a few IR receives. Trying to replicate the MakeCode micro:bit Red Light, Green Light project. 
Stoplight code:
```let REDLIGHT = 0
let GREENLIGHT = 0
let sent_State = 0
input.buttonA.onEvent(ButtonEvent.Click, function () {
    sent_State = GREENLIGHT
    light.setAll(0x00ff00)
})
input.buttonB.onEvent(ButtonEvent.Click, function () {
    sent_State = REDLIGHT
    light.setAll(0xff0000)
})
light.setPixelColor(9, 0xffff00)
sent_State = 0
GREENLIGHT = 1
REDLIGHT = 2
forever(function () {
    network.infraredSendNumber(sent_State)
})
```
Player code:

```
let moving = false
let REDLIGHT = 0
let GREENLIGHT = 0
let gameover = false
let received_State = 0
let sent_State = 0
network.onInfraredReceivedNumber(function (sent_State) {
    received_State = sent_State
    pause(__internal.__timePicker(500))
})
light.setPixelColor(0, 0x7f00ff)
received_State = 0
sent_State = 0
GREENLIGHT = 1
REDLIGHT = 2
gameover = false
forever(function () {
    if (!(gameover)) {
        if (received_State == GREENLIGHT) {
            light.setAll(0x00ff00)
        } else if (received_State == REDLIGHT) {
            light.setAll(0xff0000)
        } else {
        	
        }
        if (received_State == REDLIGHT) {
            moving = Math.abs(input.acceleration(Dimension.Strength)) - 1000 > 100
            if (moving == true) {
                gameover = true
            }
        }
    }
    if (gameover == true) {
        light.setAll(0xffff00)
    }
})
```

bborncr, 2018-11-23:
Has there been any progress on this issue?
The problem can be duplicated on a single CPX using the following code.
Just click the A button a few times and then try to play the sound clicking on B. The sound is slowed down.
``` javascript
input.buttonA.onEvent(ButtonEvent.Click, function () {
    network.infraredSendNumber(10)
})
input.buttonB.onEvent(ButtonEvent.Click, function () {
    music.playSound(music.sounds(Sounds.PowerUp))
})
```

gearsincorg, 2018-11-23:
It's extremely frustrating that there seems to be no traction on getting this issue addressed.   
It's the major stumbling block to using CPX for any kind of multi-player game.
Games are what gets kids interested in programming, and games where you compete against a friend are the most compelling.

If I knew where to look for the code for this I'd try to locate the bug, but I just can't get started.

Clearly the IR send is messing up a register or something which also used to play sounds....
It shouldn't be impossible to locate the rogue command....

Phil.

gearsincorg, 2018-11-23:
So glad to get a response.   That github link doesn't work for me, must be private.

Is there a public way to evaluate the new strategy with CPX?

Phil.

pelikhan, 2018-11-23:
https://github.com/lancaster-university/codal-samd21 Oops

pelikhan, 2018-11-23:
And we’ll take help too  :) I think we can have a branch of adafruit that uses the new Codal soon.

pelikhan, 2018-11-23:
I think ADC is the big missing thing right now.

gearsincorg, 2018-11-23:
I'm eager to do help with testing for the Adafruit CPX version.  I'm a beta tester for several FIRST robotics programs, and a  developer for the FIRST Tech Challenge SDK.  I mainly do testing, and leave the heavy coding to others more qualified.

pelikhan, 2019-03-12:
@jamesadevine let's test those scripts.

pelikhan, 2019-03-12:
Slow down is fixed on nombed.

closed 2019-03-12 by pelikhan