microsoft/pxt-adafruit #780: Degraded performance with multiple variables

pelikhan, 2018-06-18:
> I was using the AdaFruit MakeCode web experience, and noticed that this program brings the project to a crawl.
Saving/loading the project, adding blocks or any other user interactions de-grade as each variable is added to the project.

https://makecode.com/_4UfDF26RtAs2

IoTDan, 2018-06-18:
This is with the edge browser.

IoTDan, 2018-06-18:
I am seeing this in chrome, as well. changing any of the set varlables in the OnStart, for instance, is extremly slow, if the brwoser responds at all. thanks! dan

abchatra, 2018-06-18:
This is a good bug. Looks like reproes on Jacqueline machine as well. Doesn't repro on my machine.

IoTDan, 2018-06-18:
thanks! what does P2 mean in your priorities? this seems like a P1 issue, but that is on a P0 scale.

Jaqster, 2018-06-18:
Rename this file to remove the .txt
[perfprofile.diagsession.txt](https://github.com/Microsoft/pxt-adafruit/files/2113092/perfprofile.diagsession.txt)

abchatra, 2018-06-18:
This is the scale we use. 
P0->website down 
P1-> Greater than 25% of users are impacted and need a hotfix
P2->Fix before next release
P3->Likely won't fix.

pelikhan, 2018-06-18:
Javascript
```
let tenth = 0
let ninth = 0
let eighth = 0
let seventh = 0
let sixth = 0
let fifth = 0
let fourth = 0
let third = 0
let second = 0
let first = 0
first = 2
second = 0
third = 2
fourth = 2
fifth = 3
sixth = 1
seventh = 3
eighth = 3
ninth = 3
tenth = 1
forever(function () {
    if (first == 1) {
        light.setAll(0xff0000)
    }
    if (second == 1) {
        light.setAll(0xff8000)
        if (third == 1) {
            light.setAll(0xffff00)
        }
        if (fourth == 1) {
            light.setAll(0xff9da5)
        }
        if (fifth == 1) {
            light.setAll(0x00ff00)
        }
        if (sixth == 1) {
            light.setAll(0xb09eff)
        }
        if (seventh == 1) {
            light.setAll(0x00ffff)
        }
        if (eighth == 1) {
            light.setAll(0x007fff)
        }
        if (ninth == 1) {
            light.setAll(0x7f00ff)
        }
        if (tenth == 1) {
            light.setAll(0xff0080)
        }
    }
})
```

abchatra, 2018-06-18:
Doesn't repro on Chrome on Jacqueline box. Repro's on Edge consistently. Likely software rendering issue with the blocky editor. Doesn't repro on Monaco editor. Currently hypothesis is blocky rendering of if statements causes the UI thread to hog.

samelhusseini, 2018-06-18:
This issue isn't related to the use of any specific block but has more to do with the number of blocks used. 

It seems (at least on Jacqueline's machine) that it only takes about ~15 blocks for this issue to repro. I managed to repro the same issue with about 10 blocks on scratch-blocks and 15 blocks on the Blockly playground, which makes me think it's related to Edge / IE rendering all the SVG that Blockly creates. 

@IoTDan do you mind adding some hardware specs to so we can hone in on the source of this issue?

IoTDan, 2018-06-18:
Thanks ! we reproed on 3 surface books.

abchatra, 2018-06-18:
I should note, all the instances we have seen till now are related to surface Pro or surface books.

IoTDan, 2018-06-19:
@abchatra, any update here?

abchatra, 2018-06-20:
We narrowed it down to Edge+Surface issue. This happens in other products such as scratch and blockly.  Suspicion is SVG rendering bug in Edge.  Next step is to generate a etl file and get help from friends over at Edge team.

abchatra, 2018-07-17:
This is clearly a regression in Edge 42 (April udpate for windows 10)
[image]

Same issue on the same box which had Edge 41.16299  and Edge-Html 16.16299 doesn't show any performance issues.

abchatra, 2018-07-19:
Short repro: https://makecode.adafruit.com/34325-97729-56953-84285

Javscript code in question on our end: 
```javascript
Blockly.WorkspaceSvg.prototype.updateInverseScreenCTM = function() {
  var ctm = this.getParentSvg().getScreenCTM();
  if (ctm) {
    this.inverseScreenCTM_ = ctm.inverse();
  }
};
There is less than 20 calls to getScreenCTM.
```

ETL (Xperf alpha latest)
\\scratch2\scratch\abchatra\trace
You can clearly see large amount of cycles spent in the getScreenAP in BoundingRect calculations.
[image]

Other tracking issue:https://github.com/playi/pxt-wonder/issues/305

abchatra, 2018-07-23:
Edge team confirms this is fixed in RS5 and will be backported to RS4. We will wait for the fix to be ported to RS4 before closing this.

abchatra, 2018-10-03:
They fixed in their latest update. Please install the latest update (https://support.microsoft.com/en-us/help/4458469) on RS4 or install windows 10 update RS5 (Windows 10 version 1809)

closed 2018-10-03 by abchatra