microsoft/pxt-adafruit #793: Using Crickit Servo Pauses Music Playing

TheKitty, 2018-07-06:
**Describe the bug**
When using the new Crickit stepper move block, if music like the play sound block will pause while the stepper moves, then sound resumes.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Use of program attached. Adafruit Crickit + Circuit Playground Express. Build as of 7/5/2018.

**Expected behavior**
A clear and concise description of what you expected to happen.
That the stepper motor would move with a button press and any music would continue to play at the same time

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Win10 latest build
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22] Version 67.0.3396.99 (Official Build) (64-bit)

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] N/A
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Timer conflict? Or something blocking?

abchatra, 2018-07-10:
@TheKitty Can you try with pause instead of wait? Wait is blocking wait and pause gives control to other fibers.
[image]

TheKitty, 2018-07-10:
Actually steppers require a brief pause, about 10-20 milliseconds between step movements. Most of my examples have moved to 10 milliseconds.

Pause is too limited for timing, it only has 6 or so allowed values. If pause could have more control over the time settings, I could use it instead of wait.

abchatra, 2018-07-10:
You can type any value in pause. Don't have to use drop down

TheKitty, 2018-07-10:
Seriously! Now I know. I'll test, it may take a day or two. Thanks!

TheKitty, 2018-07-12:
Bug can be closed, maybe ensure wait is documented as a blocking time call? Thanks for your help.

closed 2018-07-12 by TheKitty

abchatra, 2018-07-12:
@ganicke Can we add to the documentation of wait information about it being blocking compared to pause (May be a section about comparison between pause & wait)? https://makecode.adafruit.com/reference/control/wait-micros

ganicke, 2018-07-12:
Sure, I can get at that.