microsoft/pxt-adafruit #799: No help available for Mouse Extension blocks

TheKitty, 2018-07-26:
**Describe the bug**
When you try  to pull up help for the mouse extension blocks, help is greyed out and there is no help.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
right click a Mouse block and if you click Help the right pane opens with standard block description, doc, and example(s)

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - OS: Win10
 - Browser Chrome
 - Version 67.0.3396.99 (Official Build) (64-bit)

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] n/a
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pelikhan, 2019-03-12:
@ganicke can you take a peek?

ganicke, 2019-03-12:
Sure. It needs docs for the three blocks exposed.

closed 2019-03-13 by pelikhan