microsoft/pxt-adafruit #801: The project name user inputted can't be saved

Helen09, 2018-08-01:
**Repro steps:**
1. Navigate to https://makecode.adafruit.com/beta#editor
2. Drag blocks to content pane 
3. Click "save" button on the bottom of the page 
4. Input a project name in the text box and then click "Save" 
[image]

**Expected result:**
The saved project's name should be same as the name user inputted

**Actual result:** 
The project name user inputted can't be saved
[image]

riknoll, 2018-08-07:
I've fixed this in pxt-core, just needs to be released

Helen09, 2018-11-14:
Closed as fixed