microsoft/pxt-adafruit #805: Simulator shows wrongly when tilting simulator

Helen09, 2018-08-07:
**Repro steps:**
1. Navigate to https://makecode.adafruit.com/beta#editor
2. Drag blocks as below .gif
3. Tilt simulator up or down

**Expected result:**
When tilting simulator up, it should show blue and tilting simulator down it should show yellow

**Actual result:**
When tilting simulator up, it shows yellow and tilting simulator down it shows blue
[image]

riknoll, 2018-08-07:
This is the correct behavior. The mouse is at the bottom of the simulator but the top of the board (the part with the USB) is tilting towards the user (up)

closed 2018-08-07 by riknoll

kevinjwalters, 2019-08-08:
I've rediscovered this. I think this is mainly a _naming_ issue and will continue to create confusion as is: https://github.com/microsoft/pxt-arcade/issues/1190