microsoft/pxt-adafruit #807: Building Github Packages – Critcal Error

teachmecs, 2018-08-07:
When following the instructions (https://makecode.com/packages/getting-started), I am getting 422 errors in my default JavaScript files (before I even try to add any custom code).

This error happened when I imported an existing (empty) repository from Github and when I imported a "Create New" repository through MakeCode.

[image]

(All I'm trying to do is package my really awesome binary number LED blocks!!)

riknoll, 2018-08-07:
As a workaround, you can probably edit `pxt.json` and remove `dal.d.ts` from the file list

mmoskal, 2018-08-07:
I can repro. Havn't seen it in non-CPX targets.

mmoskal, 2018-08-07:
The build should hit `/beta` in a couple minutes. Thanks @teachmecs  for reporting!

closed 2018-08-07 by mmoskal

teachmecs, 2018-08-07:
Thank you so much! It worked.

I have created a block that displays a binary number using the RGB LED ring. It's going to be SUPER helpful for our students to display sensor values and to get an EXACT readout.
(In my opinion, this block should be a standard block for MakeCode, as it's very useful.)

https://github.com/teachmecs/BinaryLEDsFinal

(As a side note, I'm absolutely not an expert with any of this, so it may not be the perfect solution. Just trying to write curriculum for our middle school students. If I really knew what I was doing, I'd pass another parameter to control the color of the LEDs.)

Next steps are to figure out how to get it approved...

teachmecs, 2018-08-08:
@riknoll @mmoskal 
Makes complete sense! Thank you both for taking the time to responde. I'm a super novice that knows just enough to get in trouble.