microsoft/pxt-adafruit #826: MakeCode for Adafruit on Windows 10 will not detect Circuit Playground Express

ChristopherGAndrews, 2018-09-10:
**Describe the bug**
MakeCode for Adafruit downloaded from the MS store will not detect my Circuit Playground Express, I get:

_Oops, we couldn't find your Adafruit Circuit Playground Express
Please make sure your Adafruit Circuit Playground Express is connected and try again._

**To Reproduce**
On Windows 10 

Steps to reproduce the behavior:
1. Launch the MakeCode for Adafruit app and start a new project
2. App any event console block to the forever loop, add a light event so you can visually know that the app is running on the device
3. Test with the similator to make sure it works as expected
4. Plug in the Adafruit Circuit Playground Express and switch into bootloader mode with the reset button
5. Click the Download button
6. Get the error, use the UF2 download to save the file to the CPLAYBOOT drive
7. Device boots with the new program, but there is no console access to the app that is running

**Expected behavior**

It detects the CPE and is able to display the output of the console in the app.

**Screenshots**

[image]

**Desktop (please complete the following information):**

OS

- Name	Microsoft Windows 10 Enterprise
- Version	10.0.15063 Build 15063

MakeCode
- Adafruit Circuit Playground Express version:  1.1.37
- Microsoft MakeCode version: 3.18.28

Adafruit Circuit Playground Express
- UF2 Bootloader v2.0.0-adafruit.5 SFHWRO
- Model: CPlay Express
- Board-ID: SAMD21G18A-CPlay-v0

abchatra, 2018-09-12:
@ChristopherGAndrews We are finding this hard to reproduce. Can you patch your board with one of the makecode .uf2 files by dragging and droping on the usb drive. They retry the same scenario? 

Does this repro on a different windows 10 device?

ChristopherGAndrews, 2018-09-17:
Apparently I did not hit the comment button, try two...

Testing on another Windwos 10 computer I was not able to reproduce the error.  I have tried two different CPEs on my computer and they both have this issue.  It seems like there is a driver issue on my computer.  I think there is a config issue the the Makecode install.

On a side note, I just found this blog post showing how to use WebUSB and Google Chrome:

https://blog.adafruit.com/2018/08/02/use-webusb-to-allow-microsoft-makecode-to-download-direct-to-circuit-playground-express/

I tested it on my computer and it works perfectly.  So all of the USB functionality is fine on my computer, it is just the method that the installed app uses to connect.  Uninstalling the app does not resolve the error.  I have tried all the steps in the Windows App troubleshooting:

https://makecode.adafruit.com/device/windows-app/troubleshoot

ChristopherGAndrews, 2018-09-17:
I have tried:

- Undocking my notebook to remove all other usb devices and hubs
- I have tried each of the three USB ports on the notebook
- I have uninstall the devices
  -  HID - HID-Compliant vendor-defined device
  -  HID - All of the "USB Input Device"
  -  USB Controllers - All of the "USB Composite Device"
  -  USB devices - PXT Device

Plugging in the PCE reinstalls all of the devices, but it still is not detected by the Windows App.

riessy11, 2021-03-31:
Hi, I am having this same exact issue - did you ever find a solution? Seems to b isolated to one laptop running windows 10 - other (older laptops) running windows 10 do not have this issue