microsoft/pxt-adafruit #829: Default volume level in set volume block too high

jedgarpark, 2018-09-19:
**Describe the bug**
The default volume of the CPX if no **set volume** block is used seems to be around level 20. The default level in the **set volume** block is 120.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a **start** block, add a **play sound power up** block
2. Upload to CPX, listen
3. Add a **set volume** block before the **play sound power up** block
4. Note how the slider and value are at the max, 128
5. Upload to CPX, listen. LOUD!

**Expected behavior**
The **set volume** block should default to 20

**Desktop (please complete the following information):**
 - OS: mac os
 - Browser chrome

abchatra, 2018-11-19:
@mmoskal Looks like you set the default volume to 128 here: https://github.com/Microsoft/pxt-common-packages/pull/467

This bug is ok to be resolved?

mmoskal, 2018-11-19:
CPX is not using the mixer package, at least not yet

abchatra, 2018-11-20:
@jedgarpark Default value, if no set volume block is 64 (and not 20). Is it ok if we set the default value in set  volume block to 64 as well?

jedgarpark, 2018-12-05:
Did this change? Currently seems default volume is zero. Try creating this as the entire script `music.playSound(music.sounds(Sounds.PowerUp))`
vs this
`music.setVolume(128)
music.playSound(music.sounds(Sounds.PowerUp))
`

samelhusseini, 2018-12-05:
This seems to indicate that the default volume is 64. see https://github.com/Microsoft/pxt-common-packages/blob/master/libs/music/music.cpp#L19

However trying this out it seems lower than that but not zero.

samelhusseini, 2018-12-05:
Okay drilled into this further. 
The default volume is 16 out of 256. This has been the case forever.
The setVolume block shows a default value of 128, the original request of this bug is to change that to be 20. or 16 to be more precise. 

As for my previous comment, I mentioned the default is 64 and so has @abchatra, that's true for what we send the synthesizer. but the setVolume block multiples it's input by 4 to match. so from a user's perspective since the volume is a range of 0 - 256, the default is 16. 

I've confirmed with @Jaqster that the default volume has always been like this. 
@jedgarpark to your issue of it being zero by default, I'd like to understand this further as I can't repro. Thanks!

jedgarpark, 2019-03-08:
Just saw this -- not sure what was going on then. Currently seems to have default that is not zero, so that's good.

abchatra, 2019-03-08:
fixed

jedgarpark, 2019-03-09:
Sorry, wasn't the issue that we were hoping to have the default value of the block read "20" and not "120"?  I don't see a change in that.

pelikhan, 2019-03-12:
Fixed by https://github.com/Microsoft/pxt-common-packages/pull/743/files

closed 2019-03-12 by pelikhan