microsoft/pxt-adafruit #830: Slow performance with the Makecode Windows App for CPX with > 10 blocks

pambrose, 2018-09-24:
**Describe the bug**
Very slow performance with the Makecode Windows App for CPX with > 10 blocks.
The same app works fine in the brwoser version.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a forever loop.
2. Add 10+ "Play note at middle C for 1/2 beat" blocks.  

After about 8 blocks, GUI gets glitchy and simulation lags.  At 12+ blocks, things grind to a halt. 
Again, the same app will work fine in the browser version.

**Expected behavior**
The Windows app should behave the same as the browser version.

**Desktop (please complete the following information):**
 - The latest Windows 10
 - Version 1.1.37 of Makecode Windows App.

**Additional context**
An example of a problematic is here: https://makecode.com/_PA29gb9UXC7T

I see that you have released a 1.1.46 version on github, but I am not sure how to install that version to see if you have already fixed this problem.

Thanks,
Paul

pelikhan, 2018-09-24:
@abchatra sounds like the Edge SVG rendering block

abchatra, 2018-10-03:
This is a bug with Edge. They fixed in their latest update.  Please install the latest update (https://support.microsoft.com/en-us/help/4458469) on RS4 or install windows 10 update RS5 (Windows 10 version 1809)

closed 2018-10-03 by abchatra