microsoft/pxt-adafruit #836: Can't build local server ! Has anything changed ?

curlyz, 2018-10-08:
**Describe the bug**
`[ 86%] Building CXX object CMakeFiles/CIRCUIT_PLAYGROUND.dir/pxtapp/base/pxt.cpp.o
[ 87%] Building CXX object CMakeFiles/CIRCUIT_PLAYGROUND.dir/pxtapp/buttons/buttons.cpp.o
[ 87%] Building CXX object CMakeFiles/CIRCUIT_PLAYGROUND.dir/pxtapp/core/codal.cpp.o
[ 88%] Building CXX object CMakeFiles/CIRCUIT_PLAYGROUND.dir/pxtapp/core/control.cpp.o

TypeError: Cannot read property 'length' of null
    at compressHex (C:\Users\curlyz\Documents\GitHub\pxt\built\pxt.js:102047:37)
    at C:\Users\curlyz\Documents\GitHub\pxt\built\pxt.js:101984:36
    at tryCatcher (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:512:31)
    at Promise._settlePromise (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:569:18)
    at Promise._settlePromise0 (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:614:10)    at Promise._settlePromises (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:693:18)    at Async._drainQueue (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\async.js:133:16)
    at Async._drainQueues (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\async.js:143:10)
    at Immediate.Async.drainQueues (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\async.js:17:14)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
{ sha: '7035d8c9f9d4d28105e7a309bc35be5f8b7751447cf9e3039301328b30db8655' }
pxt_modules/base/pxt-core.d.ts(1,1): error TS9043: The hex file is not available, please connect to internet and try again.

***
*** Build failed: target build failed: Compiler test failed
Error: Compiler test failed
    at Object.userError (C:\Users\curlyz\Documents\GitHub\pxt\built\pxt.js:98253:25)
    at C:\Users\curlyz\Documents\GitHub\pxt\built\pxt.js:132942:19
    at tryCatcher (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:512:31)
    at Promise._settlePromise (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:569:18)
    at Promise._settlePromise0 (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:614:10)    at Promise._settlePromises (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\promise.js:693:18)    at Async._drainQueue (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\async.js:133:16)
    at Async._drainQueues (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\async.js:143:10)
    at Immediate.Async.drainQueues (C:\Users\curlyz\Documents\GitHub\pxt\node_modules\bluebird\js\release\async.js:17:14)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
***`

**To Reproduce**
Steps to reproduce the behavior:
Follow exactly **Local Dev setup**

**Expected behavior**
I have previously success in building this , I remembered that the build download some pre-built hex file , it seems to be missing which caused this ?

**Screenshots**

polling C++ build [https://makecode.com/compile/7035d8c9f9d4d28105e7a309bc35be5f8b7751447cf9e3039301328b30db8655.json]
build log [https://makecode.com/compile/7035d8c9f9d4d28105e7a309bc35be5f8b7751447cf9e3039301328b30db8655.log]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 
**Additional context**
I am running Node 8.16 with NVM (if relevant)

pelikhan, 2018-10-08:
Fixing -- see https://github.com/Microsoft/pxt-adafruit/pull/835

curlyz, 2018-10-09:
Thanks @pelikhan  , it works !

closed 2018-10-09 by curlyz

pelikhan, 2018-10-09:
That branch is now in master.