microsoft/pxt-adafruit #837: bump release to pick up USB descriptor tweak in codal?

ladyada, 2018-10-08:
only change was to make USB descriptor advertise 100mA instead of 500mA current draw:
https://github.com/lancaster-university/codal-core/pull/32/files

this way HID will work on iOS devices

ladyada, 2018-10-08:
@pelikhan @mmoskal poke :)

pelikhan, 2018-10-08:
Let me get this in beta.

pelikhan, 2018-10-08:
Editor v1.1.47 will have the fix. Reopen if needed.

deanm1278, 2018-10-08:
hi @pelikhan , it looks like codal-circuit-playground is set to use this commit of codal-core https://github.com/lancaster-university/codal-core/commit/196f2947a80d4d89979c7b760341a7e8b0c6892a
according to this file in the most recent snapshot: https://github.com/lancaster-university/codal-circuit-playground/blob/master/target-locked.json

The USB descriptor change is in the master branch of codal-core and does not appear to be in 1.1.47

pelikhan, 2018-10-08:
@mmoskal @jamesadevine are we merging pkserial branch soon? Or should i pull this fix in that branch?