microsoft/pxt-adafruit #846: Tutorial "Siren" doesn't have picture on IE

Helen09, 2018-11-13:
**Repro steps:**
1. Navigate to https://makecode.adafruit.com/beta
2. Scroll down scroll bar and observe tutorial "Siren"

**Expected result:(IE)**
Tutorial "Siren" should have picture on IE
[image]

**Actual result:(Edge, Chrome & Firefox & Windows App)**
Tutorial "Siren" doesn't have picture on IE
[image]

ganicke, 2018-11-17:
OK, weird...lemme check it on IE...

ganicke, 2018-11-17:
Can't repro this, more info please:

[image]

ganicke, 2018-11-17:
Wait, this happens in the current release:

[image]

Wait I'm testing off or HEAD.

ganicke, 2018-11-17:
No, it's some IE vs. cloud problem:

[image]

Helen09, 2018-12-03:
This bug repro on below beta vesion on IE:

Adafruit Circuit Playground Express version:  1.2.9
Microsoft MakeCode version:  5.1.5
codal-circuit-playground runtime version:  v1.5.6

PR #1026 (2019-05-08): Re-encode the siren project thumb image
IE 11 didn't like something about the Siren project thumb image. Probably something to do with transparency. Anyway, I made a new siren.jpg and it seems to work now.

Fixes #846
closed 2019-05-08 by pelikhan