microsoft/pxt-adafruit #856: The example project is empty on IE

Helen09, 2018-11-15:
**Repro steps:**
1. Navigate to https://makecode.adafruit.com/beta
2. Scroll down scroll bar and open example project such as "Dice", "Energy glove", "Egg game", "patience"
3. Observe the editor content pane

**Expected result(Edge, Chrome & Firefox & Windows app):**
The example project should not be empty on IE
[image]

**Actual result:**
The example project is empty on IE
[image]

samelhusseini, 2018-11-15:
Issue here is Math.imul is not supported on IE, and we're using it in https://github.com/Microsoft/pxt/blob/28e192df12889f36015c8ed7c80c146d0b2cd0bd/pxtcompiler/emitter/hexfile.ts#L704

Fix is to polyfill imul for IE.

pelikhan, 2018-11-15:
v1.2.3 should fix thi sissue.

closed 2018-11-27 by samelhusseini